Huguet57/LIMO-Velo #10: Could you please supply some dataset?

qpc001, 2022-02-18:
Amazing jobs, could you please supply some dataset?

Huguet57, 2022-02-18:
Yay! First!

### Ouster
Try ``ouster.launch`` with this bags: [http://data.ouster.io/sample-data-1.13/index.html](http://data.ouster.io/sample-data-1.13/index.html).

### Velodyne (KITTI)
Try ``kitti.launch`` downloading data from the [KITTI Dataset](http://www.cvlibs.net/datasets/kitti/raw_data.php). It has to be the [synced+rectified data] ones. Here are a few examples:
- Residential data: easy. Slow motion, lots of features. Ex: [Run 35](https://s3.eu-central-1.amazonaws.com/avg-kitti/raw_data/2011_09_26_drive_0035/2011_09_26_drive_0035_sync.zip)
- City data: medium. More varied, lots of different scenarios. Ex: [Run 91](https://s3.eu-central-1.amazonaws.com/avg-kitti/raw_data/2011_09_26_drive_0091/2011_09_26_drive_0091_sync.zip)
- Road data: hard. Lots of moving cars, little amount of features. Ex: [Run 15](https://s3.eu-central-1.amazonaws.com/avg-kitti/raw_data/2011_09_26_drive_0015/2011_09_26_drive_0015_sync.zip)

These are zips that can be transformed to ROS bags thanks to [kitti2bag](https://github.com/tomas789/kitti2bag).

Huguet57, 2022-02-18:
Also, consider the bags given by LIO-SAM at [https://github.com/TixiaoShan/LIO-SAM#sample-datasets](https://github.com/TixiaoShan/LIO-SAM#sample-datasets). These are very slow motion, so it should have no problem. All bags that can be scenarios that can be solved by LIO-SAM, can be solved with LIMO-Velo.

Huguet57, 2022-02-18:
### Fast-LIO dataset examples
Also consider the data given by Fast-LIO in [https://github.com/hku-mars/FAST_LIO#4-rosbag-example](https://github.com/hku-mars/FAST_LIO#4-rosbag-example). They give Velodyne bags and Livox Avia bags.

Today I might add support for Livox and create a specific ``avia.launch`` to run those.

closed 2022-02-20 by qpc001