Huguet57/LIMO-Velo #11: Build error

arihantb2, 2022-02-22:
I am building LIMO-Velo on my machine

Host OS: Ubuntu 18.04
CPU Architecture: x86-64
Build tool - catkin tools

The error I am getting when building from a clean workspace is,
```
Errors     << limovelo:make /path/to/ws/logs/limovelo/build.make.003.log                                                                                                                                                                                                                                                                                                            
/path/to/ws/src/limo-velo/src/Modules/Mapper.cpp: In member function ‘Matches Mapper::match(const State&, const Points&)’:
/path/to/ws/src/limo-velo/src/Modules/Mapper.cpp:47:26: error: expected ‘=’ before ‘:’ token
             for (Point p : points) {
                          ^
/path/to/ws/src/limo-velo/src/Modules/Mapper.cpp:53:13: error: expected ‘;’ before ‘return’
             return matches;
             ^~~~~~
/path/to/ws/src/limo-velo/src/Modules/Mapper.cpp:53:13: error: expected primary-expression before ‘return’
/path/to/ws/src/limo-velo/src/Modules/Mapper.cpp:54:9: error: expected primary-expression before ‘}’ token
         }
         ^
/path/to/ws/src/limo-velo/src/Modules/Mapper.cpp:54:9: error: expected ‘)’ before ‘}’ token
/path/to/ws/src/limo-velo/src/Modules/Mapper.cpp:54:9: error: expected primary-expression before ‘}’ token
/path/to/ws/src/limo-velo/src/Modules/Mapper.cpp:47:13: error: invalid controlling predicate
             for (Point p : points) {
             ^~~
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-dev’
make[2]: *** [CMakeFiles/limovelo.dir/src/Modules/Mapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/limovelo.dir/all] Error 2
make: *** [all] Error 2
```

Looks like a trivial build issue but I am not able to find the solution myself. Thank you for your help @Huguet57

arihantb2, 2022-02-22:
Looks like this answer on stackoverflow solves it.
Link: https://stackoverflow.com/questions/33493215/openmp-error-invalid-controlling-predicate

Huguet57, 2022-02-22:
Seems like OpenMP doesn't like ``!=`` in the for loop definitions. And the colon operator is equivalent to ``for(auto it = deviceList.begin(); it != deviceList.end(); ++it)``...

Very interesting, this should be an easy fix. Thank you for reporting! :)

Huguet57, 2022-02-22:
After 8 minutes of compilation (...) in principle this commit: [81de64c](https://github.com/Huguet57/LIMO-Velo/commit/81de64ce86a27c53930bb2b9ce4270f3531b92bb) should be it. Can you try it now?

closed 2022-02-22 by Huguet57

arihantb2, 2022-02-23:
Yes, it works now! Thank you