Huguet57/LIMO-Velo #14: Something wrong after commit: e6bebf5b87a8869d683bc0a921fa49f455ea2c1a

qpc001, 2022-02-24:
When I start to play the bag, LIMO-Velo run a few scan (n<5), then stuck.

Huguet57, 2022-02-24:
Does e6bebf5b87a8869d683bc0a921fa49f455ea2c1a work?

qpc001, 2022-02-24:
> Does [e6bebf5](https://github.com/Huguet57/LIMO-Velo/commit/e6bebf5b87a8869d683bc0a921fa49f455ea2c1a) work?

no, but [db67b8e285e13882a89d8ad78437f7678e12e30e] work well.

Huguet57, 2022-02-24:
Does it happen with ``xaloc.launch``?

qpc001, 2022-02-24:
> Does it happen with `xaloc.launch`?

I run `test.launch` with my own dataset.

Huguet57, 2022-02-24:
:alert: I just found that e6bebf5b87a8869d683bc0a921fa49f455ea2c1a has a bug. ``fill_config()`` needs ``nh`` as argument. Try a later commit

Huguet57, 2022-02-24:
However, I just tried it and it doesn't compile... Does your error happen in latter commits?

qpc001, 2022-02-24:
> However, I just tried it and it doesn't compile... Does your error happen in latter commits?

yes, I can upload my dataset if you need.

Huguet57, 2022-02-24:
Interesting. It seems that is a time management related error. If you could upload your dataset that would be great!

Huguet57, 2022-02-24:
Does your LiDAR scan have "time" data?

Huguet57, 2022-02-24:
Actually I think I might just found the error, try to comment "heuristic" part in the YAML file and uncomment the "without heuristic"

qpc001, 2022-02-24:
> 

This work, is there a bug ?

Huguet57, 2022-02-24:
Your LiDAR must go @ 10Hz and if t2-t1 is less than 0.1 seconds then it doesn't find points with timestamps between t1 and t2

qpc001, 2022-02-24:
yes, I use vlp-16

Huguet57, 2022-02-24:
Mm this is really interesting, a modern Velodyne ROS driver should output 'time' value in the points. Could you upload a couple of seconds of data to see what's happening?

Huguet57, 2022-02-24:
[This post](https://github.com/ros-drivers/velodyne/issues/281) is related to this problem.

Does your ``/velodyne_points`` look like this in RViz when you put ``Channel Transform = "Intensity"`` and ``Channel Time = time``? It should look like a rainbow, just like this image.

[image]

Huguet57, 2022-02-24:
If it does, then it would really be interesting to get to look at the data.

qpc001, 2022-02-24:
I just sent you a 15s ros bag.

An interesting things is , my velodyne point cloud time stamp is the end of the scan.

Huguet57, 2022-02-24:
> 

> An interesting things is , my velodyne point cloud time stamp is the end of the scan.

This is usually the case for Velodyne LiDARs, and not having 'time' data is dangerous since then you will always go 0.1 seconds desynchronized with the IMU. LIMO-Velo assumes there's time data in each point and this goes from ``0`` (earliest point) to ``0.1`` (latest point). Hence the time of beginning without time is badly calculated here:

https://github.com/Huguet57/LIMO-Velo/blob/c70c311ef05d9f046debd1b00f42fc89977f3a99/src/Utils/PointCloudProcessor.cpp#L42-L45

And it is indeed the "finish" time.

I will have a look at the data you sent me, thank you!

Huguet57, 2022-02-24:
[image]

This is really interesting, your data does have ``time`` data (I hope you don't mind showing the screenshot) but it goes from ``-0.1`` to ``0`` (actually, it makes way more sense that way). This calls for a big fix, I will get onto it.

Huguet57, 2022-02-24:
@qpc001 replacing this line:
https://github.com/Huguet57/LIMO-Velo/blob/c70c311ef05d9f046debd1b00f42fc89977f3a99/src/Utils/PointCloudProcessor.cpp#L44
For:
`return Conversions::microsec2Sec(pcl.header.stamp);`

Already improves a lot your case. I am going to find a way to detect automatically or at least choose in the YAML the 'time' field format.

Huguet57, 2022-02-24:
@qpc001 issue fixed, in your VLP-16 case, the new parameters should be:

https://github.com/Huguet57/LIMO-Velo/blob/f5ab92ed0bb1f54f0ad581e25b9cd06b1666c11c/config/params.yaml#L25-L28

As updated in the ``params.yaml`` file.

closed 2022-02-24 by Huguet57