Huguet57/LIMO-Velo #20: Commit [08c7093] has better accuracy than [main]

Huguet57, 2022-02-28:
> 我明白你现在的意思了。确实，现在它可以避免从 null 到 initialized 的跳转。
> 
> 我会进入它。

for more test, I think commit [08c7093cc7212106a996f60ee5f4691d20e3f7f0] is better than branch [main]. I mean , the accuracy and fluency.

_Originally posted by @qpc001 in https://github.com/Huguet57/LIMO-Velo/issues/12#issuecomment-1051441887_

Huguet57, 2022-03-02:
- Commit [08c7093cc7212106a996f60ee5f4691d20e3f7f0] is better than main.
- Commit [c70c311ef05d9f046debd1b00f42fc89977f3a99] keeps its quality.

Huguet57, 2022-03-02:
I have not found any differences between the quality of [08c7093cc7212106a996f60ee5f4691d20e3f7f0] vs. [main]. Leaving this open for some days until someone says otherwise.

qpc001, 2022-03-13:
> [我没有发现 [ 08c7093](https://github.com/Huguet57/LIMO-Velo/commit/08c7093cc7212106a996f60ee5f4691d20e3f7f0) ] 与 [main]的质量有任何区别。将其打开几天，直到有人另有说明。

Maybe I can send you my bag file to test.

Huguet57, 2022-03-14:
That would be interesting, yes!

qpc001, 2022-03-15:
> 
A bag file has been sent to your google mail.