Huguet57/LIMO-Velo #22: How soon the [hdmap] branch will be release?

qpc001, 2022-03-17:


Huguet57, 2022-03-17:
There's some good news/bad news about the [hdmaps] branch.

Good news is that 90% of the work is done. Anyone can try it now and it will likely work.

Bad news is that it depends on the 9-DOF support since initial state is also "non-zero".

So the HD-map will be ready as soon as I figure out how to do the 9-DOF support cleanly. I would expect within a week if everything goes well.

qpc001, 2022-03-18:
If i know the init pose, the [hdmaps] branch will work?

Huguet57, 2022-03-19:
Well, it actually works now as well but there are the same bugs as the 9-DOF.

When those are fixed, 9-DOF and HD-maps will be available and working 100% of the times.