Huguet57/LIMO-Velo #26: Leaf size is too small for the input dataset

wpwebrobotics, 2022-07-12:
Hi, thanks for sharing the code.
I am trying the slam using a rosbag made from Velodyne VLP-16 output and IMU Vectornav 100. The rosbag seems to me to have nothing special but I get this error:
[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.
I have tried different values for downsample_prec (from 0.1 to 3) to no avail.
What could it be? Thank you.

Huguet57, 2022-07-19:
Check the number of points in `compensated`. This is a problem that happens when using the `pcl::VoxelGrid` and a too small pointcloud.

You can try bigger deltas (try keeping it at 0.1 in the last stage of the `Initialization`), lowering the `downsample_rate` (to 1?) and check that the `min_dist` is adequate for your robot/aplication (maybe indoor robots do need points of the 4 meter radius) ideally the `min_dist` would cover just only the robot.

If it's a puntual occurrence, you could try to increase the `Config.MAX_POINTS2MATCH` hyperparameter (not referenced in the YAML):
https://github.com/Huguet57/LIMO-Velo/blob/57f32058eddab9ed1461f856bb3987742c22f362/src/main.cpp#L81

Huguet57, 2022-07-19:
I would guess it's likely to be the `min_dist` value if it's an indoor application. The default value of 4 in the `xaloc.yaml` file is for a 2m x 1.5m autonomous car in an outdoor scenario.

Huguet57, 2023-04-16:
We have observed that this problem disappears if we require more IMU readings when initializing. Thanks for pointing it out and sorry for the delayed response!

closed 2023-04-16 by Huguet57