Huguet57/LIMO-Velo #28: compensator assert killing the process

bertaveira, 2023-01-14:
I have been converting this to ROS2 and noticed that the assert in the compensator upsample function was killing the process. Upon further investigation, I think this assert is a bug. 
Line: https://github.com/Huguet57/LIMO-Velo/blob/main/src/Modules/Compensator.cpp#L72

As it says in the "path" function the states are between "just before t1 to t2" and the imus are between "first state to just after t2".

Then in the assert we check if `imus.front().time <= states.front().time` which should always throw an error since the oldest imu should not be older than the older state. 

I had to comment this line and then it works.

Huguet57, 2023-01-14:
Weird! As seen in lines 108-112 of the Accumulator initialization: hhttps://github.com/Huguet57/LIMO-Velo/blob/57f32058eddab9ed1461f856bb3987742c22f362/src/Modules/Accumulator.cpp#L108-L112

It's supposed to initialize when enough IMU samples have been gathered - and then set `initial_time` (which in turn, sets `initial_state`'s time).

bertaveira, 2023-01-14:
But upsample is called by the `Compensator::path()` which will only get a portion of IMUs and States. 

If I am not mistaken this piece of code ensures that the oldest state passed to the `Compensator::upsample()` is older than the older IMU passed to it as well (when doing `states.push_front(accum.get_prev_state(t1));`). It also ensures that the latest IMU is later than the latest State
https://github.com/Huguet57/LIMO-Velo/blob/57f32058eddab9ed1461f856bb3987742c22f362/src/Modules/Compensator.cpp#L36-L49

It fells like this assert should have the first inequality changed from `<=` to `>=`
https://github.com/Huguet57/LIMO-Velo/blob/57f32058eddab9ed1461f856bb3987742c22f362/src/Modules/Compensator.cpp#L71-L73

Huguet57, 2023-01-14:
Now I see that the `get` method gets from `t1` to `t2` included:
https://github.com/Huguet57/LIMO-Velo/blob/543d92edcd6ac1e4b5bd1ea9b55ddd280b274b24/include/Headers/Accumulator.hpp#L69
So, in principle, `states.front().time` and `imus.front().time` should be the same - if the state has been derived from an IMU timestamp (which I think always happens). Weird that it crashed your code.

I would vote against changing the inequality and more in favor of just removing (?) the first part. The idea was actually that the IMU stream spanned the State stream because then we can get a continuous upsampling (no large chunk from first state to first IMU).

It is more interesting to investigate why it's crashing the code. For now, commenting it, doesn't seem to be dangerous.